J2Kbr/TitanTwo #297: X52 Input mapping refinement request

thievery80, 2021-04-05:
Hi there!

Dear J2KBR,
fantastic work with the new HID for the X52!
As i understand, it seems to be hard to add all physical Inputs, so i would like to request a change of the HID to maximise the usage. Please find table underneath.
Reason for: we have 3 rotary/slider which are not really much usefull, as XB/PS does have limited analogs, which are in use already in favour of other functions (also to be as close to the gamepad mapping as possible). Therefore it would make sense to switch over and refine it to:

Name | old HID | new HID
-- | -- | --
Mouse Click 1 | BUTTON_14 | BUTTON_14
Mouse Click 2 | POINT_2_Y | POINT_2_Y
Mouse UP/DOWN | Paddle_4 | POINT_1_X
Mouse LEFT/RIGHT | Paddle_3 | POINT_1_Y
Mouse Scroll UP |   | ACCEL_1_X
Mouse Scroll DOWN |   | ACCEL_1_Y
POV HAT 3 UP | BUTTON_5 | BUTTON_5
POV HAT 3 RIGHT | BUTTON_4 | BUTTON_4
POV HAT 3 DOWN | BUTTON_8 | BUTTON_8
POV HAT 3 LEFT | BUTTON_7 | BUTTON_7
POV HAT 2 UP | ACCEL_1_X | Paddle_1
POV HAT 2 RIGHT | ACCEL_2_X | Paddle_2
POV HAT 2 DOWN | ACCEL_1_Y | Paddle_3
POV HAT 2 LEFT | ACCEL_1_Z | Paddle_4
POV Hat 1 UP | BUTTON_10 | BUTTON_10
POV Hat 1 RIGHT | BUTTON_13 | BUTTON_13
POV Hat 1 DOWN | BUTTON_11 | BUTTON_11
POV Hat 1 LEFT | BUTTON_12 | BUTTON_12
Trigger 1 | BUTTON_6 | BUTTON_6
Trigger2 | Paddle_1 |  
Pinky Switch |   | POINT_2_X
Fire A | GYRO_1_Y | GYRO_1_Y
Fire B | GYRO_1_X | GYRO_1_X
Fire C | BUTTON_9 | BUTTON_9
Fire | BUTTON_21 | BUTTON_21
Button E | BUTTON_17 | BUTTON_17
Rotary 2 | POINT_1_Y |  
Button D | BUTTON_15 | BUTTON_15
Button I | BUTTON_16 | BUTTON_16
Rotary 1 | POINT_1_X |  
Precision Slider | POINT_2_X |  
Function | BUTTON_2 | BUTTON_2
Start/Stop | BUTTON_1 | BUTTON_1
Reset | BUTTON_3 | BUTTON_3
Joystick X Axis | STICK_1_X | STICK_1_X
Joystick Y Axis | STICK_1_Y | STICK_1_Y
Jouytick Z Axis | STICK_2_X | STICK_2_X
Throttle Axis | STICK_2_Y | STICK_2_Y
Toggle 1 | GYRO_1_Z | ACCEL_1_Z
Toggle 2 | ACCEL_2_Z | ACCEL_2_X
Toggle 3 | BUTTON_18 | ACCEL_2_Y
Toggle 4 | BUTTON_19 | BUTTON_18
Toggle 5 | ACCEL_2_Y | BUTTON_19
Toggle 6 | BUTTON_20 | BUTTON_20
Mode Switch 1 | PADDLE_2 | GYRO_1_Z
Mode Switch 2 |   | ACCEL_2_Z
Mode Switch 3 |   |  (calulated of Switch1+2)

for the new physical inputs coming in:
- we saw already the Mode Switches (sure you are aware of)
- Pinky Switch: this one was mapped at the very initial generic HID
- Mouse Scroll UP/Down: here i hope for your magic to find & bind them :)

As you also saw, i did dome changes in PADDLE´s and some random to get a more consistent HID for the joystick (optional)

Many thanks in advance!!!

thievery80, 2021-04-05:
check [forum entry]( https://www.consoletuner.com/forum/viewtopic.php?f=27&t=17966)

thievery80, 2021-04-06:
here´s a filter on only changing channels:

Name | old HID | new HID
-- | -- | --
Mouse UP/DOWN | Paddle_4 | POINT_1_X
Mouse LEFT/RIGHT | Paddle_3 | POINT_1_Y
Mouse Scroll UP | unused | ACCEL_1_X
Mouse Scroll DOWN | unused | ACCEL_1_Y
POV HAT 2 UP | ACCEL_1_X | Paddle_1
POV HAT 2 RIGHT | ACCEL_2_X | Paddle_2
POV HAT 2 DOWN | ACCEL_1_Y | Paddle_3
POV HAT 2 LEFT | ACCEL_1_Z | Paddle_4
Trigger2 | Paddle_1 | unused
Pinky Switch | unused | POINT_2_X
Rotary 2 | POINT_1_Y | unused
Button D | BUTTON_15 | BUTTON_16
Button I | BUTTON_16 | BUTTON_15
Rotary 1 | POINT_1_X | unused
Precision Slider | POINT_2_X | unused
Toggle 2 | ACCEL_2_Z | ACCEL_2_X
Toggle 3 | BUTTON_18 | ACCEL_2_Y
Toggle 4 | BUTTON_19 | BUTTON_18
Toggle 5 | ACCEL_2_Y | BUTTON_19
Toggle 6 | BUTTON_20 | BUTTON_20
Mode Switch 1 | PADDLE_2 | ACCEL_1_Z
Mode Switch 2 | unused | ACCEL_2_Z
Mode Switch 3 | (calculated by: | Switch 1+2)

phillowAT, 2021-05-15:
with a colleague, we figured out a remapping/re-asignment that should work out for both version (and hopefully all platforms).
Please find the proposal here: https://www.consoletuner.com/forum/download/file.php?id=4242

based on the collaboration work done here: https://www.consoletuner.com/forum/viewtopic.php?f=27&t=17720&p=111857&hilit=X52#p111546

It would be great to test the behavior upfront without putting it into the next update right away for a POC.

Many thanks in advance!