J2Kbr/TitanTwo #300: X-52 Pro

ModifiedSoul, 2021-04-10:
- GTuner does not recognise Full functionality of all available Axis and buttons
- Throttle lights are not working

Thanks

phillowAT, 2021-05-15:
hi! pls check #297, where we tried to get botch versions fixed for all platforms

ModifiedSoul, 2021-05-17:
Nice, I'll delete this soon

Just wondering but are using the mini stick inputs ? 
On mine they appear as paddle 3 and paddle 4 

 I ask as I think it would make a natural head movement control

ModifiedSoul, 2021-05-17:
> hi! pls check #297, where we tried to get botch versions fixed for all platforms

Nice, I'll delete this soon

Just wondering but are using the mini stick inputs ?
On mine they appear as paddle 3 and paddle 4

I ask as I think it would make a natural head movement control

phillowAT, 2021-05-17:
Hi there!
from an INPUT perspective, there is no issue to go ahead and control "headlook" with the ministick.
Problem is: on e.g. XB, you would need to have 2 analog axis (floating int) to get that covered, which needs to have dedicated/direct input, such as X/Y/Z stick axis. As we are lacking of channel-offerings from console, i tend to use the method with an "activator" button and DPADS, where you reset the headlook position on button release