J2Kbr/TitanTwo #307: Firm 1.11.30: Logitech G27 contantly disconnecting/reconnecting

mannylouey2013, 2021-06-21:
TL;DR: After updating my Titan 2 to the latest firmware, my wheel, which worked perfectly previously, and still works perfectly on PC and PS3, started disconnecting during gameplay.

This issue appeared after upgrading, I used the wheel almost daily before upgrading. The wheel will work perfectly fine for a minute or so until suddenly, all at once, the following happens:
- Force feedback in the wheel goes limp in my hands
- The PS4 and game will report that the controller is disconnected
- Almost immediately, the wheel will reconnect and begin calibration. The PS4 will display the 'who's using this controller?' display.
- The issue will occur within 30~ seconds of gameplay. Leaving the controller alone does not reproduce the issue.

Setup:
- T2 plugged into PS4 via Output port. Output set to Logitech racing wheel.
- G27 in port A. DS4 in port B.
- Bluetooth adapter installed in T2, but issue occurs even after I removed it.
- No 7.5v power adapter, but issue occurs even when I add one.

Troubleshooting:
- Ensured no macros, input mappings or any custom adjustments of any kind were present on the Titan 2.
- Tried removing bluetooth module and adding power.
- Tested wheel on PC and PS3. Works fine just fine.
- Restarted PS4

Please help! I would really appreciate the option to downgrade in the interim as I have an FIA race which I really need to earn points on that's coming up.

mannylouey2013, 2021-06-22:
Video of issue here
https://streamable.com/vfyals

Ksathral, 2021-09-23:
I have had similar issues in the past, with my controllers being disconnected continiously both via bluetooth and usb A / B input. 
You did connectan external power adapter and the issue still occurs for you, so I don't know if this will work for you.

I know this is not the intended way the prog port should be used, but when I connect my phone charger, or a usb c cable connected to the same powered usb hub that my titantwo is connected to on my ps4, then the issue with controllers being disconnected stopped.
I do not have a power adapter that fits on the titan two, which is the reason I tried this and somehow it worked for me.