J2Kbr/TitanTwo #324: Xbox Adaptive Controller: Won't connect via BT if unit NOT marked "Made in China"

oneswitch, 2022-03-31:
I've found with two Xbox Adaptive Controllers that are NOT marked "Made in China" that they won't connect via BlueTooth to the XAC. I'm calling these problem controllers the "Dublin" controllers, as they may have been made in Dublin, Ireland (I'm guessing).

There is a different in the descriptor files (Attached).

If this can be fixed, it would really help some Disabled gamers to have a more comfortable playing expeirience. Hope you can put your genius to it soon, Jefferson.

Thanks.

Barrie
OneSwitch.org.uk

[XAC Model 1836 - 02560030776826 Dublin.txt](https://github.com/J2Kbr/TitanTwo/files/8387846/XAC.Model.1836.-.02560030776826.Dublin.txt)
[XAC Model 1836 - 02560033817929 Made in China.txt](https://github.com/J2Kbr/TitanTwo/files/8387847/XAC.Model.1836.-.02560033817929.Made.in.China.txt)

oneswitch, 2022-04-02:
I've also heard from another user, that this may be Xbox Series S/X related as they're struggling to connect with a "Made in China" unit. Tricky to pin this one down.

oneswitch, 2022-04-05:
This is the input descriptor for the XAC (remote from me) that the owner can't get to connect via BT. I see all three of these are different. I'm told this is from a fully updated XAC.

[XAC Model - Beau.txt](https://github.com/J2Kbr/TitanTwo/files/8421326/XAC.Model.-.Beau.txt)

beau76, 2022-05-04:
Hi Jefferson ,
I was able to run a test with a xac with firmware 4.5.168.0 and the Bluetooth on the titan two works but if you have  version 5.13.3143.0  it does not it would be a big help if you could fix this as I was unable to rollback Beau